TILOS-AI-Institute/MacroPlacement #10: Stack trace when running FormatTranslators test2.py

braamBeresford, 2022-07-12:
When trying to run partition_design as part of CodeElements/FormatTranslators/test/test2.py I receive a stack trace. Cannot find source of issue.

Command:
```set top_design ariane  
set def_file ./design/ariane.def  
set ALL_LEFS "  
    ./lefs/NangateOpenCellLibrary.tech.lef  
    ./lefs/NangateOpenCellLibrary.macro.mod.lef  
    ./lefs/fakeram45_256x16.lef  
"  
foreach lef_file ${ALL_LEFS} {  
    read_lef $lef_file  
}  
read_def ./design/ariane.def   
partition_design -max_num_inst 2000000 -min_num_inst 40000 -max_num_macro 12 -min_num_macro 4 -net_threshold 20  -virtual_weight 50 -num_hop 0 -timing_weight 1000 -report_file ${top_design}.hgr   
exit  
```

Error:
```OpenROAD GITDIR-NOTFOUND 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: ./lefs/NangateOpenCellLibrary.tech.lef
[INFO ODB-0223]     Created 22 technology layers
[INFO ODB-0224]     Created 27 technology vias
[INFO ODB-0226] Finished LEF file:  ./lefs/NangateOpenCellLibrary.tech.lef
[INFO ODB-0222] Reading LEF file: ./lefs/NangateOpenCellLibrary.macro.mod.lef
[INFO ODB-0225]     Created 135 library cells
[INFO ODB-0226] Finished LEF file:  ./lefs/NangateOpenCellLibrary.macro.mod.lef
[INFO ODB-0222] Reading LEF file: ./lefs/fakeram45_256x16.lef
[INFO ODB-0225]     Created 1 library cells
[INFO ODB-0226] Finished LEF file:  ./lefs/fakeram45_256x16.lef
[INFO ODB-0127] Reading DEF file: ./design/ariane.def
[INFO ODB-0128] Design: ariane
[INFO ODB-0097] 		Created 100000 Nets
[INFO ODB-0130]     Created 495 pins.
[INFO ODB-0131]     Created 87585 components and 578948 component-terminals.
[INFO ODB-0132]     Created 2 special nets and 175170 connections.
[INFO ODB-0133]     Created 113121 nets and 403778 connections.
[INFO ODB-0134] Finished DEF file: ./design/ariane.def
Running Partition Design...
[INFO PAR-0402] Traversed logical hierarchy
	Number of std cell instances: 0
	Total area: 0.0
	Number of hard macros: 0
Stack trace:
 0# 0x0000000000C15A16 in openroad
 1# 0x00007FEA8BA99400 in /lib64/libc.so.6
 2# sta::LibertyCell::isBuffer() const in openroad
 3# par::AutoClusterMgr::getBufferNetUtil(odb::dbBlock*, std::vector<std::pair<odb::dbNet*, odb::dbNet*>, std::allocator<std::pair<odb::dbNet*, odb::dbNet*> > >&) in openroad
 4# par::AutoClusterMgr::getBufferNet() in openroad
 5# par::AutoClusterMgr::partitionDesign(unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, bool, char const*, char const*, float, float, float, float) in openroad
 6# par::PartitionMgr::partitionDesign(unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, bool, char const*, char const*, float, float, float, float) in openroad
 7# 0x00000000018ADCD6 in openroad
 8# 0x00007FEA8E5E1EB2 in /lib64/libtcl8.5.so
 9# 0x00007FEA8E62636C in /lib64/libtcl8.5.so
10# TclObjInterpProcCore in /lib64/libtcl8.5.so
11# 0x00007FEA8E5E1EB2 in /lib64/libtcl8.5.so
12# 0x00007FEA8E62636C in /lib64/libtcl8.5.so
13# 0x00007FEA8E62E647 in /lib64/libtcl8.5.so
14# TclEvalObjEx in /lib64/libtcl8.5.so
15# 0x00007FEA8E66927F in /lib64/libtcl8.5.so
16# 0x00007FEA8E5E1EB2 in /lib64/libtcl8.5.so
17# 0x00007FEA8E62636C in /lib64/libtcl8.5.so
18# 0x00007FEA8E62E647 in /lib64/libtcl8.5.so
19# TclEvalObjEx in /lib64/libtcl8.5.so
20# 0x00007FEA8E5E91D0 in /lib64/libtcl8.5.so
21# 0x00007FEA8E5E1EB2 in /lib64/libtcl8.5.so
22# 0x00007FEA8E62636C in /lib64/libtcl8.5.so
23# 0x00007FEA8E62E647 in /lib64/libtcl8.5.so
24# TclEvalObjEx in /lib64/libtcl8.5.so
25# 0x00007FEA8E5ECF00 in /lib64/libtcl8.5.so
26# 0x00007FEA8E5E1EB2 in /lib64/libtcl8.5.so
27# 0x00007FEA8E62636C in /lib64/libtcl8.5.so
28# TclObjInterpProcCore in /lib64/libtcl8.5.so
29# 0x00007FEA8E5E1EB2 in /lib64/libtcl8.5.so
30# 0x00007FEA8E62636C in /lib64/libtcl8.5.so
31# TclObjInterpProcCore in /lib64/libtcl8.5.so
32# 0x00007FEA8E5E1EB2 in /lib64/libtcl8.5.so
33# 0x00007FEA8E5E2F1E in /lib64/libtcl8.5.so
34# Tcl_EvalEx in /lib64/libtcl8.5.so
35# Tcl_Eval in /lib64/libtcl8.5.so
36# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
37# ord::tclAppInit(Tcl_Interp*) in openroad
38# Tcl_Main in /lib64/libtcl8.5.so
39# main in openroad
40# __libc_start_main in /lib64/libc.so.6
41# 0x0000000000C125F7 in openroad
```

LEF and DEF collaterals:
https://github.com/TILOS-AI-Institute/MacroPlacement/tree/main/CodeElements/FormatTranslators/test/LefDef2ProtocolBufferFormat

ZhiangWang033, 2022-07-15:
Hi braamBeresford,  Thanks for  opening this issue.   I wonder how did you run our code.    I have doubled check the code on my end and I did not have any problems.   To run the lef/def translator, you just need to go to 
${your_dir}/CodeElements/FormatTranslators/test/LefDef2ProtocolBufferFormat and type "python test2.py".     If you have more questions,  please let me know.     Thanks, Zhiang.

braamBeresford, 2022-07-17:
ZhiangWang,
I have try the step you decribe (cd into Lef2DefProtocolBufferFormat -> python test2.py). 

This is with Ubuntu18.04.6 LTS and python3.6.9  .
Can you tell me your OS configuration and how you installed necessary dependencies?

korstu, 2022-07-18:
Hello, 
I have one question. Could you please tell me how can I create these files by using only DEF/LEF files. 
ariane.hgr
ariane.hgr.fix
ariane.hgr.instance
ariane.hgr.io
ariane.hgr.macro_pin
ariane.hgr.net
ariane.hgr.outline 

In [Lef2DefProtocolBufferFormat -> python test2.py], without these files, it will have some errors
 
Thank you.

ZhiangWang033, 2022-07-19:
Hi,

You can check the openroad website for all the dependencies: https://github.com/The-OpenROAD-Project/OpenROAD.
If you can run openroad app binary (already provided in the my repo),  then you should be able to see this:   
[image]

ZhiangWang033, 2022-07-21:
Please let me know if I can close this issue.

cr145, 2022-07-22:
I'm having a similar issue. 

I am not able run the provided openroad binary as I get errors related to libraries (for some libraries I was unable to install the correct version). While I'm doing all of this inside a docker container, I think the issue may be due to differences in kernel version. Could you provide the kernel version of the system you are using (in addition to the ubuntu and python versions)?

To get around this, I compiled openroad inside the docker container. When I run it I get a stack trace similar to the one posted by @braamBeresford.

[image]

braamBeresford, 2022-07-27:
I haven't resolved it. As @cr145 said, when trying to use your provided binary I get a library error with Qt5.11. I'm still trying to resolve this one.
I also compiled my own binary of OpenROAD and got the errro dump out from above.
What OS and Kernel are you using?

ZhiangWang033, 2022-07-27:
Hi,  could you please check your module load file ?   I attached my module load file.
[image]

ZhiangWang033, 2022-07-27:
Here is my os version
[image]

diaoerlie, 2022-09-22:
same issue , ubuntu 20.04 python 3,8

diaoerlie, 2022-09-23:
[image]

diaoerlie, 2022-09-23:
[INFO ODB-0134] Finished DEF file: ./design/ariane.def
Running Partition Design...
[INFO PAR-0402] Traversed logical hierarchy
	Number of std cell instances: 0
	Total area: 0.0
	Number of hard macros: 0
Stack trace:
 0# 0x000056339BDEBD11 in ../../src/utils/openroad
 1# 0x00007F32EFEAF090 in /lib/x86_64-linux-gnu/libc.so.6
 2# sta::LibertyCell::isBuffer() const in ../../src/utils/openroad
 3# par::AutoClusterMgr::getBufferNetUtil(odb::dbBlock*, std::vector<std::pair<odb::dbNet*, odb::dbNet*>, std::allocator<std::pair<odb::dbNet*, odb::dbNet*> > >&) in ../../src/utils/openroad
 4# par::AutoClusterMgr::getBufferNet() in ../../src/utils/openroad
 5# par::AutoClusterMgr::partitionDesign(unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, bool, char const*, char const*, float, float, float, float) in ../../src/utils/openroad
 6# par::PartitionMgr::partitionDesign(unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, bool, char const*, char const*, float, float, float, float) in ../../src/utils/openroad
 7# 0x000056339CCA9923 in ../../src/utils/openroad
 8# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
 9# 0x00007F32F1B9C924 in /lib/x86_64-linux-gnu/libtcl8.6.so
10# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
11# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
12# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in ../../src/utils/openroad
13# ord::tclAppInit(Tcl_Interp*) in ../../src/utils/openroad
14# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
15# main in ../../src/utils/openroad
16# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
17# _start in ../../src/utils/openroad
[INFO]  Error!  /home/*/Desktop/MacroPlacement-main/CodeElements/FormatTranslators/test/LefDef2ProtocolBufferFormat/rtl_mp/ariane.hgr.io  does not exist!!!

ZhiangWang033, 2022-09-27:
Hi,  Could you please build your own OpenROAD (https://github.com/ravi-varadarajan/OpenROAD.git) binary ?

diaoerlie, 2022-09-28:
hi zhi ang, it works!

closed 2022-09-28 by ZhiangWang033