TILOS-AI-Institute/MacroPlacement #11: How to create .hgr.* in [CodeElements/Clustering/test/rtl_mp/] folder

korstu, 2022-07-18:
Hello,

I have one question. Could you please tell me how can I create these files by using only DEF/LEF files.

ariane.hgr
ariane.hgr.fix
ariane.hgr.instance
ariane.hgr.io
ariane.hgr.macro_pin
ariane.hgr.net
ariane.hgr.outline

In [Lef2DefProtocolBufferFormat -> python test2.py], without these files, it will have some errors

Thank you.

ZhiangWang033, 2022-07-19:
Hi,

You do not need to provide these files.   All you need to do is to go to CodeElements/FormatTranslators/test/LefDef2ProtocolBufferFormat directory and type "python test2.py".
You will see logs as following
[image]

korstu, 2022-07-19:
No. I mean that if I use my own DEF, LEF files, how should I create these files similar to.

my_own.hgr
my_own.hgr.fix
my_own.hgr.instance
my_own.hgr.io
my_own.hgr.macro_pin
my_own.hgr.net
my_own.hgr.outline
 
In your code, you have given these files already.

ZhiangWang033, 2022-07-21:
Sorry for the misunderstanding.   If you are going to run your own design,  please refer to this directory:  https://github.com/TILOS-AI-Institute/MacroPlacement/tree/main/CodeElements/CodeFlowIntegration.

Running "python flow.py" will automatically generate the clustered netlist in protocol buffer format.

ZhiangWang033, 2022-07-21:
Please let me know if I can close the issue.

closed 2022-09-27 by ZhiangWang033

jason-nguyentronghieu, 2022-10-11:
I thought some of you might use all newer versions than Ubuntu 18.04, and the file **`openroad`** in bin folder might have been built on Ubuntu 18.04. That’s reason why some of you might face this issue. I suggest to do

1.  Install `Qt5`
```
# change a path to save .run file
cd /path/to/
wget https://download.qt.io/new_archive/qt/5.11/5.11.2/qt-opensource-linux-x64-5.11.2.run
chmod +x qt-opensource-linux-x64-5.11.2.run
./qt-opensource-linux-x64-5.11.2.run

# copy all libs from `Qt5.11.2` to `/usr/lib/x86_64-linux-gnu/ `
sudo cp -a /path/to/Qt5.11.2/5.11.2/gcc_64/lib/. /usr/lib/x86_64-linux-gnu/
```
2. Install `libreadline7` and `libtinfo5`
```
wget http://archive.ubuntu.com/ubuntu/pool/main/n/ncurses/libtinfo5_6.1-1ubuntu1_amd64.deb
sudo dpkg -i libtinfo5_6.1-1ubuntu1_amd64.deb

wget http://archive.ubuntu.com/ubuntu/pool/main/r/readline/libreadline7_7.0-3_amd64.deb
sudo dpkg -i libreadline7_7.0-3_amd64.deb
```
3. Install Python 3.6
```
sudo apt-get update
sudo add-apt-repository ppa:deadsnakes/ppa
sudo apt-get update
sudo apt install python3.6
sudo apt-get install libpython3.6-dev
```
4. The other problem might related to file `hmetis` in **`/src/utils/hmetis`**
    When we run `./hmetis`, if the output as below
    ```bash
    ./hmetis
    bash: ./hmetis: No such file or directory
    ```
    It means the current OS doesn’t recognize this built file. To solve this issue, we have to install some libs as below:
    ```bash
    sudo apt-get install libc6:i386
    sudo apt-get install libstdc++6:i386
    ```
    After installing, if we run  `./hmetis` again, the output should be like this:
    ```bash
    ./hmetis
    ********************************************************************************
    Usage of hMetis:  
    [Option1]:  hmetis HGraphFile Nparts UBfactor
    [Option2]:  hmetis HGraphFile FixFile Nparts UBfactor
    [Option3]:  hmetis HGraphFile Nparts UBfactor Nruns CType RType Vcycle Reconst dbglvl
    [Option4]:  hmetis HGraphFile FixFile Nparts UBfactor Nruns CType RType Vcycle Reconst dbglvl
    [INFO] HGraphFile : 
    [INFO] FixFile : 
    [INFO] Nparts : 2
    [INFO] UBfactor : 5
    [INFO] Nruns : 10
    [INFO] CType : 1
    [INFO] RType : 1
    [INFO] Vcycle : 1
    [INFO] Reconst : 0
    [INFO] dbglvl : 16
    ```

After checking all issues above, I think that you can run `python test2.py` smoothly without any problems.

I hope it will be useful for some of you facing these issues