TILOS-AI-Institute/MacroPlacement #24: ../Grouping/src/utils/openroad: undefined symbol: _ZdlPvm, version Qt_5

garumaru, 2022-09-22:
Hello,

I followed the steps "https://github.com/The-OpenROAD-Project/OpenROAD" to setup openroad environment, however when running "python3 flow.py" under MacroPlacement/MacroPlacement-main/CodeElements/CodeFlowIntegration path, I will get below error:

../Grouping/src/utils/openroad ../Grouping/src/utils/extract_hypergraph.tcl setup.tcl
../Grouping/src/utils/openroad /ic/proj/comet/root/MacroPlacement/MacroPlacement-main/CodeElements/CodeFlowIntegration/extract_hypergraph.tcl
../Grouping/src/utils/openroad: symbol lookup error: ../Grouping/src/utils/openroad: undefined symbol: _ZdlPvm, version Qt_5

The OS is Centos 8. 

Anyone knows how to solve this issue? 
Thank you very much!

ZhiangWang033, 2022-09-23:
@sharkoo7,  Thanks for your reply.  Yes,  sharkoo7 is right.  I suggest you take a look at OpenROAD installation guide.

garumaru, 2022-09-27:
thank you guys @ZhiangWang033 @sharkoo7 , I switch the OS to ubuntu18.04 and install tools as I did in centos8, and it works

closed 2022-09-27 by garumaru

jason-nguyentronghieu, 2022-10-11:
Mainly, this issue might happen on the newer versions than Ubuntu 18.04 such as: Ubuntu 20.04. To fix this issue without switching to Ubuntu 18.04, we can use these steps below.

# change a path to save .run file
cd /path/to/
wget https://download.qt.io/new_archive/qt/5.11/5.11.2/qt-opensource-linux-x64-5.11.2.run
chmod +x qt-opensource-linux-x64-5.11.2.run
./qt-opensource-linux-x64-5.11.2.run

# copy all libs from `Qt5.11.2` to `/usr/lib/x86_64-linux-gnu/ `
sudo cp -a /path/to/Qt5.11.2/5.11.2/gcc_64/lib/. /usr/lib/x86_64-linux-gnu/