TILOS-AI-Institute/MacroPlacement #26: Can FormatTranslators generate .plc file？

sharkoo7, 2022-09-22:
Google circuit training require both `initial.plc` and `.pb.txt`  .But Running test2.py only generates pb.txt file. In test2.py, you set  `output_file = design + ".plc"` , but seems like it is not used.

ZhiangWang033, 2022-09-23:
Yes,  you are right.  We are working on enabling this function.   I will ping you when the function is online.

sharkoo7, 2022-09-27:
Thank you so much for working on it! I enabled this function based on your code in the past few days. And the generated `initial.plc` file seems to work fine in the circuit training process. I would love share it, if you are interested.

ZhiangWang033, 2022-09-27:
Hi  Sharkoo7,  Thanks for your work.  We have enabled this function also.

closed 2022-09-28 by ZhiangWang033

RaidToRadar, 2023-06-06:
I would like this feature. would you mind sharing this code with me please