TILOS-AI-Institute/MacroPlacement #28: gridding.py No such file or directory: 'Desktop/MacroPlacement-main/CodeElements/Gridding/test/rtl_mp/ariane.hgr.outline'

diaoerlie, 2022-09-27:
hi ,

i've run up-to-dated macroplacement flow (add code flow), 
openroad is install and pass recursive check in my machine, and openroad (setup.tcl) is pointed to my local build path. 

I found error in:
gridding = GriddingLefDefInterface(gridding_src_dir, design, setup_file, tolerance, halo_width, openroad_exe)  line 297
self.ExtractInputs() line 345

error  is :
FileNotFoundError: [Errno 2] No such file or directory: '/home/*/Desktop/MacroPlacement-main/CodeElements/CodeFlowIntegration/rtl_mp/ariane.hgr.outline'

could you kindly help me check this issue?

diaoerlie, 2022-09-27:
here's file generated in rtl_mp
 
 positon.txt
 io_net.txt
ariane.hgr.blockage
top_glue_logic_macro_cluster_0_part_0.txt.block
ariane.hgr.net
ariane.hgr.block
top_glue_logic_macro_cluster_0_part_0.txt.net

diaoerlie, 2022-09-27:
by the way , i've used Desktop/MacroPlacement-main/CodeElements/Gridding/test/test_gridding.py 
the error is the same

ZhiangWang033, 2022-09-27:
Hi diaoerlie,  I think it's related to the openroad binary.  Since we have got many feedbacks about OpenROAD.  Could you do a git pull and build your own OpenROAD binary (https://github.com/ravi-varadarajan/OpenROAD.git),  then retry ?

diaoerlie, 2022-09-28:
hi zhi ang,

I've build openroad locally from public git : https://github.com/The-OpenROAD-Project/OpenROAD
do you mean, i should use this git?  https://github.com/ravi-varadarajan/OpenROAD.git

ZhiangWang033, 2022-09-28:
Yes,  the codes are different.

diaoerlie, 2022-09-28:
thanks i will try it

diaoerlie, 2022-09-28:
it's seems work, let's wait and see

diaoerlie, 2022-09-28:
hi zhi ang,

here's what we generated, after python3 flow.py: 

ariane.hgr.vertex
ariane.hgr.outline
ariane.hgr
ariane.hgr.part.637

my question is : how to combine (or feed) this output into P&R tools (openroad / innovus)

thank you so much

ZhiangWang033, 2022-09-28:
No.   You cannot do this.  the code elements are used to replicate the gridding/grouping/clustering.   I am not clear what you want.   Do you want to run Circuit Training ?  Or do you want to run Gridding/Grouping/Clustering ?

diaoerlie, 2022-09-29:
i‘d like to use circuit training, and use CT's output feed into openroad / innovus tools and run place and route.

closed 2022-09-29 by diaoerlie

sakundu, 2022-09-29:
@diaoerlie 
Just for your reference:
You can use the [plc_pb_to_placement_tcl.py](https://github.com/TILOS-AI-Institute/MacroPlacement/blob/main/Flows/util/plc_pb_to_placement_tcl.py) script to generate the macro_place.tcl file. This you can source in the innovus shell to generate the macro placement. 

Here is an example of generating the macro_place.tcl from the final *.plc file and the protobuf netlist.
`python plc_pb_to_placement_tcl.py rl_opt_placement.plc netlist.pb.txt macro_place.tcl`

garumaru, 2022-09-29:
> Hi diaoerlie, I think it's related to the openroad binary. Since we have got many feedbacks about OpenROAD. Could you do a git pull and build your own OpenROAD binary (https://github.com/ravi-varadarajan/OpenROAD.git), then retry ?

a suggestion that you might need to change the description of "git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD.git" to "git clone --recursive https://github.com/ravi-varadarajan/OpenROAD.git"

:)