TILOS-AI-Institute/MacroPlacement #38: question abount CT run?

diaoerlie, 2022-10-17:
hi,

i am following your guide in : plc_client, 

I have confuse in two point:
#1. replace the environment.py with environment_ct.py  --> do you mean:  mv  environment.py  environment_ct.py 
"Then, copy plc_client_os.py under the same directory"  --> really confuse what does it mean?

#2. here's logs from test bench , is it normal?  and what is this flow generate?

022-10-17 20:15:29.456503: I tensorflow/stream_executor/cuda/cuda_diagnostics.cc:156] kernel driver does not appear to be running on this host (comet-Precision-3630-Tower): /proc/driver/nvidia/version does not exist
############################ TEST PROXY COST ############################
overlap_threshold default 0.004
#[PLC FILE FOUND] Loading info from .plc file
#[ROUTES PER MICRON] Hor: 10.00, Ver: 10.00
#[CONGESTION SMOOTH RANGE] Smooth Range: 2
#[PLACEMENT GRID] Col: 35, Row: 33
#[INFO WIRELENGTH] Matched Wirelength cost -- GL 0.05018660769988099, OS 0.05018660769390266
#[INFO DENSITY] Matched density cost -- GL 0.7500617702325654, OS 0.7500617686422556
#[INFO CONGESTION] Matched congestion cost -- GL 2.311353997750248, OS 2.3113539969146424
                  +++++++++++++++++++++++++++++
                  +++ TEST PROXY COST: PASS +++
                  +++++++++++++++++++++++++++++

thank you so much for this question

Dinple, 2022-10-17:
Hi,

1. If you have already downloaded circuit_training from Google's Github, you can run our code by replacing circuit_training/environment/environment.py with environment_ct.py (please keep the naming to be the same as environment.py for function calls). Then, you will need to move the plc_client_os.py into the same folder under circuit_training/environment/ (no need to change name or anything). Now, you can start your run as usual and circuit_training should work with our open-source code.
2. This is normal. These logs you see are individual tests to find out discrepencies in the output. In most of the tests, you will manually input these canvas information and point to your test file locations. This particular test you are running is testing the three proxy cost used by circuit_training. As you can see, we have a matching result. To toggle different tests, you can go to the bottom of the testbench and uncomment any tests you would like to see. We are also happy to know if you have found any unmatching results.

Thanks for checking out our code. I will make an update in the future to make these instructions a bit more clear.

Dinple, 2022-10-21:
Closed for inactivity. Reopen if needed.

closed 2022-10-21 by Dinple