TILOS-AI-Institute/MacroPlacement #40: Files not found when converting lef/def to protocol buffer format

gbrown40, 2022-10-28:
When I try to run MacroPlacement/CodeElements/FormatTranslators/test/LefDef2ProtocolBufferFormat/test2.py to convert lef/defs to the protocol buffer format, I get the error: 

[INFO]  Error!  /MacroPlacement/CodeElements/FormatTranslators/test/LefDef2ProtocolBufferFormat/rtl_mp/ariane.hgr.io  does not exist!!!

Looking in the repo, I do not see a rtl_mp folder anywhere.

jason-nguyentronghieu, 2022-10-29:
Hello @gbrown40,

Try to build this repo (https://github.com/ravi-varadarajan/OpenROAD) from the closed issue before and re-run `test2.py` again from **MacroPlacement/CodeElements/FormatTranslators/test/LefDef2ProtocolBufferFormat/test2.py** with a built openroad 

I hope it is helpful for you

diaoerlie, 2022-10-30:
hi gbrown, i am a also a user, you may find some tips from close issue

jason-nguyentronghieu, 2022-11-01:
Hi @gbrown40 

I thought you might use all newer versions than Ubuntu 18.04, and the file **`openroad`** in bin folder might have been built on Ubuntu 18.04. That’s reason why some of you might face this issue. I suggest to do
 
1.  Install `Qt5` (_if there is an error relating to undefined symbol: _ZdlPvm, version Qt_5_)

```
# change a path to save .run file
cd /path/to/
wget https://download.qt.io/new_archive/qt/5.11/5.11.2/qt-opensource-linux-x64-5.11.2.run
chmod +x qt-opensource-linux-x64-5.11.2.run
./qt-opensource-linux-x64-5.11.2.run

# copy all libs from `Qt5.11.2` to `/usr/lib/x86_64-linux-gnu/ `
sudo cp -a /path/to/Qt5.11.2/5.11.2/gcc_64/lib/. /usr/lib/x86_64-linux-gnu/
```
2. Install `libreadline7` and `libtinfo5` (_if there is an error relating to libreadline7,libtinfo5_)
```
wget http://archive.ubuntu.com/ubuntu/pool/main/n/ncurses/libtinfo5_6.1-1ubuntu1_amd64.deb
sudo dpkg -i libtinfo5_6.1-1ubuntu1_amd64.deb

wget http://archive.ubuntu.com/ubuntu/pool/main/r/readline/libreadline7_7.0-3_amd64.deb
sudo dpkg -i libreadline7_7.0-3_amd64.deb
```
3. Install Python 3.6 (_if there is an error related to Python 3.6_)
```
sudo apt-get update
sudo add-apt-repository ppa:deadsnakes/ppa
sudo apt-get update
sudo apt install python3.6
sudo apt-get install libpython3.6-dev
```
4. The other problem might related to file `hmetis` in **`/src/utils/hmetis`**
    When we run `./hmetis`, if the output as below
    ```bash
    ./hmetis
    bash: ./hmetis: No such file or directory
    ```
    It means the current OS doesn’t recognize this built file. To solve this issue, we have to install some libs as below:
    ```bash
    sudo apt-get install libc6:i386
    sudo apt-get install libstdc++6:i386
    ```
    After installing, if we run  `./hmetis` again, the output should be like this:
    ```bash
    ./hmetis
    ********************************************************************************
    Usage of hMetis:  
    [Option1]:  hmetis HGraphFile Nparts UBfactor
    [Option2]:  hmetis HGraphFile FixFile Nparts UBfactor
    [Option3]:  hmetis HGraphFile Nparts UBfactor Nruns CType RType Vcycle Reconst dbglvl
    [Option4]:  hmetis HGraphFile FixFile Nparts UBfactor Nruns CType RType Vcycle Reconst dbglvl
    [INFO] HGraphFile : 
    [INFO] FixFile : 
    [INFO] Nparts : 2
    [INFO] UBfactor : 5
    [INFO] Nruns : 10
    [INFO] CType : 1
    [INFO] RType : 1
    [INFO] Vcycle : 1
    [INFO] Reconst : 0
    [INFO] dbglvl : 16
    ```

After checking all issues above, I think that you can run `python test2.py` smoothly without any problems.

I hope it will be useful for some of you facing these issues

gbrown40, 2022-11-02:
@jason-nguyentronghieu 
I built my own openroad binary, so I don't think that's an issue. The issue seems to be that I do not have needed files (like ariane.hgr.io) in the rtl_mp folder. Looking through other git issues, it seems like for the Ariane design these files should already be in the repo (as opposed to needing to create them for other designs), but I do not see them.

jason-nguyentronghieu, 2022-11-02:
@gbrown40 
There is only one way if you do want to have these needed files (like ariane.hgr.io) in the **rtl_mp** folder. In the original repo (OpenROAD), it wouldn't generate files such as _.hgr.io, .hgr.net, .hgr.io.instance, .hgr.macro_pin, etc_,  if you built your own openroad binary. Therefore, you have to use the given [openroad](https://github.com/TILOS-AI-Institute/MacroPlacement/blob/main/CodeElements/FormatTranslators/src/utils/openroad). 
If you use the given [openroad](https://github.com/TILOS-AI-Institute/MacroPlacement/blob/main/CodeElements/FormatTranslators/src/utils/openroad), then [my comment](https://github.com/TILOS-AI-Institute/MacroPlacement/issues/40#issuecomment-1297994880) above might be helpful for you.

gbrown40, 2022-11-03:
@jason-nguyentronghieu 
Is there anyway to use your own openroad exe? (For instance, in flow.py is says "You need to specify your own openroad exe"). I was able to resolve some of the issues with using the given openroad exe with your comment, but I am getting errors when trying to install Qt5 that I haven't been able to fix.

jason-nguyentronghieu, 2022-11-03:
@gbrown40 
In the closing issue, there is a link of a customized [OpenROAD](https://github.com/ravi-varadarajan/OpenROAD.git). So you can build your own OpenROAD binary for only [CodeElements/CodeFlowIntegration/flow.py](https://github.com/TILOS-AI-InstituteMacroPlacement/blob/main/CodeElements/CodeFlowIntegration/flow.py). But you need to uncomment some lines( from **1967** to **2343**)  in [autocluster.cpp](https://github.com/ravi-varadarajan/OpenROAD/blob/master/src/par/src/autocluster.cpp). 

For running this one [CodeElements/FormatTranslators/test/LefDef2ProtocolBufferFormat/test2.py](https://github.com/TILOS-AI-Institute/MacroPlacement/blob/main/CodeElements/FormatTranslators/test/LefDef2ProtocolBufferFormat/test2.py), we have to use their given [openroad](https://github.com/TILOS-AI-Institute/MacroPlacement/blob/main/CodeElements/FormatTranslators/src/utils/openroad). 

**Which errors are you getting related to Qt5?** 
I'm gonna give you some tips relating to Qt5

1. Install QT5
```
cd /path/to/
wget https://download.qt.io/new_archive/qt/5.11/5.11.2/qt-opensource-linux-x64-5.11.2.run
chmod +x qt-opensource-linux-x64-5.11.2.run
./qt-opensource-linux-x64-5.11.2.run
```
2. Run the given [openroad](https://github.com/TILOS-AI-Institute/MacroPlacement/blob/main/CodeElements/FormatTranslators/src/utils/openroad) for testing your own build openroad by this command 
```
# make sure to install Qt5 first
export LD_LIBRARY_PATH=/home/your_name/Qt5.11.2/5.11.2/gcc_64/lib/ && openroad -gui
```
3. After checking, if it works, you can insert the command above in your python file such as _**FormatTranslators.py**_

```
def CreateODB(self):
        file_name = "generate_odb.tcl"
        line  = "set top_design " + self.design + "\n"
        line += "set def_file " + self.def_file + "\n"
        line += 'set ALL_LEFS "\n'
        for lef in self.lef_list:
            line += '    ' + lef + '\n'
        line += '"\n'
        line += 'foreach lef_file ${ALL_LEFS} {\n'
        line += '    read_lef $lef_file\n'
        line += '}\n'
        line += "read_def " + self.def_file + "\n"
        line += "partition_design -max_num_inst 2000000 -min_num_inst 40000 "
        line += "-max_num_macro 12 -min_num_macro 4 "
        line += "-net_threshold 20  -virtual_weight 50 "
        line += "-num_hop 0 -timing_weight 1000 "
        line += "-report_file ${top_design}.hgr \n"
        line += "exit\n"
        f = open(file_name, "w")
        f.write(line)
        f.close()

        cmd = "export LD_LIBRARY_PATH=/home/your_name/Qt5.11.2/5.11.2/gcc_64/lib/ && " +  self.openroad_exe + ' ' + file_name
        os.system(cmd)

        cmd = "rm " + file_name
        os.system(cmd)
```

gbrown40, 2022-11-04:
@jason-nguyentronghieu 
when I run `./qt-opensource-linux-x64-5.11.2.run`, I get the error
`QXcbConnection: Could not connect to display`
`Aborted (core dumped)`

jason-nguyentronghieu, 2022-11-04:
@gbrown40 
I have never seen this error. If it is possible, how about trying to install  Ubuntu on a virtual machine to test first. Because I guess your current Ubuntu has some problems.

gbrown40, 2022-11-16:
@jason-nguyentronghieu 
Still getting the same error after trying on a virtual machine with Ubuntu installed.

jason-nguyentronghieu, 2022-11-17:
Hello @gbrown40 
After installing a new Ubuntu OS, there are some steps needed to do first:

1. Download [OpenROAD](https://github.com/The-OpenROAD-Project/OpenROAD.git)

This step will help you install all dependencies
```bash
git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD.git
cd OpenROAD
# installs dependencies to compile the openroad binary
sudo  ./etc/DependencyInstaller.sh -dev 
```
2. Download [MacroPlacement](https://github.com/TILOS-AI-Institute/MacroPlacement.git)

Try to check whether this given `openroad` works or not

```bash
git clone --recursive https://github.com/TILOS-AI-Institute/MacroPlacement.git
cd MacroPlacement/CodeElements/FormatTranslators/src/utils
./openroad -gui
```

3. If there are any errors, try to check suggestions above 

 I hope it will be helpful for you.

ZhiangWang033, 2023-04-08:
Hello @gbrown40,

Did you solve your problem ?   If you still have the problem,  please let us know.

closed 2023-04-08 by ZhiangWang033