TILOS-AI-Institute/MacroPlacement #67: Are the ISPD23 paper benchmark placements available?

profmadden, 2023-07-25:
Hi -- I'm working on macro placement, and would like to take a look at the placements from the ISPD '23 paper (mostly as a sanity check on how I'm computing HPWL and other metrics).  Ideally, I'd love to see the placements from RePlACe, as well as the RL and annealing implementations.

Are there bookshelf/lef-def files available somewhere?  I'd prefer the final placements to actually running the code (the code can be a moving target with updates, and I don't have the CPU cycles easily available for things like RL).

sakundu, 2023-07-26:
Dear @profmadden,

Could you please provide more specifics regarding the test cases you're interested in? Are you referring to modern benchmarks like [Ariane](https://github.com/TILOS-AI-Institute/MacroPlacement/tree/main/Flows/NanGate45/ariane133/scripts/cadence), [BlackParrot](https://github.com/TILOS-AI-Institute/MacroPlacement/tree/main/Flows/NanGate45/bp_quad/scripts/cadence), and [MemPool Group](https://github.com/TILOS-AI-Institute/MacroPlacement/tree/main/Flows/NanGate45/mempool_group/scripts/cadence) on the Nangate45 enablement? If so, scripts available at the linked locations can be utilized to generate the placement of the respective testcases using the commercial flow.

Are you interested in viewing the final solutions from RePlAce, RL, and SA? If so, I can share the placed def file with you. I need to verify whether sharing the run details is permissible under our tool licensing agreement, but I'd be happy to do so if possible.

Which tool are you planning to use to generate the final placement considering you already the have the macro placement solution? In the ISPD23 context, we utilized Innovus 21.1 to generate the final placement. For RePlAce and SA, I can provide the necessary inputs and scripts for you to execute on your end. However, to produce the placed database, Innovus 21.1 will be required. Also, note that for SA and RePlAce solutions, we leverage the `refine_macro_placement` command in Innovus to legalize the macros.

Thanks,
Sayak

profmadden, 2023-07-26:
I'd like the final placements for anything and everything, if possible.  I have results from my bisection based placer feng shui for the ICCAD04 benchmarks -- it looks like my HPWL numbers from 20 years ago beat CT and SA on average, and even beats RePlAce on one benchmark(!).  There's something funky going on with IBM10 in particular, and I'd love to get to the bottom of it.

DEF files for the modern designs with the various placers would be great too.  I don't have easy access to the commercial flows, and it would take a lot of time and effort to get everything running and produce the placements.  The final placements are really what I'd like to look at.  I don't need the flow working, and don't need any details on any commercial script or tool configuration.

My placement flow is entirely with my own tools -- feng shui did unified macro and standard cell placement back in '04, and I've got a tweak or two since then that improves results.  I can load in the final placements into the current feng shui build -- that'll let me know for sure that I'm calculating wire length in the same way.

profmadden, 2023-08-05:
@sakundu -- any word on this?  The ICCAD04 benchmarks are probably the higher priority; I'd really like to figure out what's happening with IBM10.

profmadden, 2023-08-27:
@sakundu -- wanted to ping you again -- the ICCAD04 benchmark placements would be super useful, if you can make them available!

profmadden, 2024-10-04:
@sakundu -- ping again!