TILOS-AI-Institute/MacroPlacement #70: invalid command name "partition_design"

esonghori, 2023-09-15:
Hi, 

I am trying to use LEF/DEF to proto format code and I am getting the following error:

```
Error: generate_odb.tcl, 12 invalid command name "partition_design"
```

I am compiling openroad myself, since I get don't have python3.6 on my machine and can't use the openroad binary you provided [here](https://github.com/TILOS-AI-Institute/MacroPlacement/tree/main/CodeElements/FormatTranslators/src/utils). I was wondering if the openroad you provided is modified, since I couldn't find any documentation for `partition_design` in openroad documentation.

Thank you.

luarss, 2023-09-18:
@esonghori They are using an older version of the OpenROAD binary I think. OpenROAD has shifted to using new commands for partitioning [1](https://openroad.readthedocs.io/en/latest/main/src/par/README.html). 

Alternatively, I think you can also try an older version of OpenROAD (around Jun'22) and compile from scratch.

esonghori, 2023-09-18:
Thank you @luarss for the information. Do you know why they used partitioning for translating the LEF/DEF files into proto files. I don't understand why partitioning is needed for translation.

esonghori, 2023-09-19:
I tried replacing the `partition_design` command with `triton_part_design` in the code, however, the flags do not match perfectly from the old command to the new one.

We used to have: 

``` 
partition_design -max_num_inst 2000000 -min_num_inst 40000
  -max_num_macro 12 -min_num_macro 4
  -net_threshold 20  -virtual_weight 50
  -num_hop 0 -timing_weight 1000
  -report_file ${top_design}.hgr
```

I tried the following command instead: 
```
triton_part_design -solution_file ${top_design}.hgr -num_parts 1 -timing_aware_flag false \n
```

However, I get the following error in the logs:
```
========================================
[STATUS] Starting TritonPart Partitioner
========================================
[INFO PAR-0168] [INFO] Partitioning parameters**** 
[INFO PAR-0102] Number of partitions = 2
[INFO PAR-0016] UBfactor = 1
[INFO PAR-0017] Seed = 1
[INFO PAR-0018] Vertex dimensions = 1
[INFO PAR-0019] Hyperedge dimensions = 1
[INFO PAR-0020] Placement dimensions = 0
[INFO PAR-0021] Timing aware flag = true
[INFO PAR-0103] Guardband flag = false
[INFO PAR-0023] Global net threshold = 1000
[INFO PAR-0024] Top 1000 critical timing paths are extracted.
[INFO PAR-0025] Fence aware flag = false
[INFO PAR-0030] Solution file = ariane.hgr
========================================
[STATUS] Reading netlist**** 
[STATUS] Extracting timing paths**** 
[INFO PAR-0178] maximum_clock_period : 0 second
[INFO PAR-0179] normalized extra delay : inf
[INFO PAR-0180] We normalized the slack of each path based on maximum clock period
[INFO PAR-0181] We normalized the slack of each net based on maximum clock period
[STATUS] Finish traversing timing graph
[WARNING PAR-0138] Reset the slack of all unconstrained hyperedges to 0 seconds
[ERROR PAR-2677] There is no vertices and hyperedges
```
It looks like the `triton_part_design` command didn't find the design. Any idea how to fix this?

luarss, 2023-09-22:
Hi @esonghori the new API assumes you have read a design either via netlist or hypergraph file. There are test cases here, hope it helps: https://github.com/The-OpenROAD-Project/OpenROAD/tree/65aff1f6e4163316511ae2a3266ca058f150c3e2/src/par/test

diaoerlie, 2023-12-26:
hello @esonghori  ,  i have the same problem here, is it sovled?

diaoerlie, 2023-12-27:
@esonghori hello, read_lib is must for partiton_design