TILOS-AI-Institute/MacroPlacement #73: Questions about changing the initial set of placement locations.

cmnkkk, 2023-09-25:
Hello, after reading the question about initial set of placement locations (from physical synthesis) affect the (relative) quality of the CT.[Question 1](https://tilos-ai-institute.github.io/MacroPlacement/Docs/OurProgress/#Question1) I am curious how the vacuous placements file was generated.

Are they generated directly by changing the coordinate information within the [file](https://github.com/google-research/circuit_training/tree/main/circuit_training/environment/test_data/ariane)(circuit_training/environment/test_data/ariane)? Or generated in some other unmentioned way?

wponghiran, 2023-09-25:
I believe the method used to change coordinate does not matter as it will be the resulted `plc` will be similar. 
Either 1 or 2 should end up with the same `plc` that later consumes by CT.
1. Use physical design tool to relocate all blocks and feed to `FormatTranslator` to generate `plc` file
2. Use python script to rewrite `plc` file

cmnkkk, 2023-10-02:
Thank you very much for your answer.

As a student in internship, I still have doubts about pre-trained model. When I finish training the [Ariane model](https://github.com/google-research/circuit_training/tree/main/circuit_training/environment/test_data/ariane) from physical synthesis, I get the output folder. If I want to use the pre-trained model at this point, can I just keep the policies file and change the input file?

Also in my programme, only the checkpoints in the policies folder are updated multiple times. Does this contain the information I need for the pre-trained model?