TILOS-AI-Institute/MacroPlacement #75: AttributeError error when running Simulated Annealing

keyplay, 2024-03-12:
Hi, when I executed the following cmd, there is an AttributeError. May I know how to solve this? 
```
python sa_multicore.py
```
[image]

Thanks for your help.

NJUer211300063, 2024-03-20:
You should set the right path to plc_wrapper_main binary in plc_client.py(you should build your plc_wrapper_main binary at first）
Here is my setting:
[image]

keyplay, 2024-05-12:
In SA.py, use plc_client and placement_util in Plc_client folder rather than CT
```
sys.path.append('../Plc_client/')
import plc_client_os as plc_client
import placement_util_os as placement_util
```

closed 2024-05-12 by keyplay